yguitton/CP-Seeker #38: Anglais

ronan-cariou, 2024-03-21:
Il doit  rester un « importation » dans le code pour information à l’utilisateur qu’il faut remplacer par « import » tout court.
Idem « exportation » en « export ».

DamienCode404, 2024-03-25:
Avec la ligne de commande suivante : 

`findstr /s /i /m "mot_a_chercher" *.*`

- `/s` : Recherche récursive dans les sous-dossiers.
- `/i` : Recherche insensible à la casse (le mot peut être en majuscules ou en minuscules).
- `/m` : Affiche uniquement les noms des fichiers contenant le motif.
- `"mot_a_chercher"` : Le mot que l'on souhaite rechercher.
- `*.*` : Les étoiles servent à prendre en compte toutes les chaines de cratères en nom de fichier et en extension de fichier.

[image]

Je vais donc modifier le mot "importation" dans le fichier "server\file.R" par "import" mais le mot "exportation" n'est pas présent dans les fichiers de codes "modifiables".

ronan-cariou, 2024-03-25:
Ok je regarderai le résultat

ronan-cariou, 2024-08-01:
Validé pour v2.1.3. J'archive l'issue.

commit 5d5b4f17 (2024-03-22): Anglais

Closes #38
closed 2024-08-01 by ronan-cariou