yguitton/CP-Seeker #41: Temps de process

ronan-cariou, 2024-03-25:
Utiliser les balises horaires (print(Sys.time()) en début et fin de déconvolution (fichier process.R) et d'export (fichier process_results.R).pou calculer et renseigner par différence la durée de process dans la cellule dédiée de l’onglet de l’export Excel. Ca implique de conserver les valeurs dans la DB je pense..

DamienCode404, 2024-03-26:
Test des ajouts dans l'export du fichier excel :

[image]

Possibilité d'ajouter d'autres informations sur le CPU, la mémoire ...
Je n'ai pas encore stocké ses informations dans la base de données. 

Le script permettant de trouver le modèle du PC de l'utilisateur est en PowerShell donc il ne fonctionnera pas avec un autre système d'exploitation que Windows.

DamienCode404, 2024-03-27:
Ajout d'une nouvelle section dans la feuille Excel "Parameters" dédié pour la récupération des composants de la machine de l'utilisateur.

[image]

DamienCode404, 2024-04-10:
Erreur lorsque l'utilisateur lance une déconvolution, quitte et relance le logiciel et qu'il demande directement un export Excel. La précédente matrice est enregistrée, mais les informations de temps de process ne sont pas stockées entre les sessions. 

<p align="center">
[image]
</p>

DamienCode404, 2024-04-16:
Proposition d'une nouvelle table SQLite pour garder en mémoire pour chaque déconvolution les info : 

```
CREATE TABLE "daconvolution_info" (
	"daconvolution_info"	INTEGER NOT NULL UNIQUE,
	"time_start"	TEXT,
	"time_end"	TEXT,
	"time_diff"	TEXT,
	"computer_manufacturer"	TEXT,
	"computer_model"	TEXT,
	"os_info"	TEXT,
	"system_type"	TEXT,
	"cpu_manufacturer"	TEXT,
	"processor_info"	TEXT,
	"cpu_cores"	TEXT,
	"cpu_speed"	TEXT,
	"memory_info"	TEXT,
	"memory_speed"	TEXT,
	PRIMARY KEY("daconvolution_info" AUTOINCREMENT)
);
```

jsaintvanne, 2024-04-23:
For me ok in https://github.com/yguitton/CP-Seeker/pull/47

BUT miss the loading bar I think during the process

ronan-cariou, 2024-08-01:
Validé pour v2.1.3. J'archive l'issue.