yguitton/CP-Seeker #42: Error log

ronan-cariou, 2024-03-29:
Faire en sorte que le error log ne soit pas écrasé à chaque réouverture de CP-Seeker. Il s'agirait de nommer le fichier "error_XXX" avec XXX donnant la date-heure-minute de création du fichier, i.e. à l'ouverture d'une session.

DamienCode404, 2024-03-29:
Test du nouveau fichier de log avec la date au format jour-mois-année_heure-minutes :

[image]

DamienCode404, 2024-04-04:
[image]

jsaintvanne, 2024-04-23:
For me ok in https://github.com/yguitton/CP-Seeker/pull/47

ronan-cariou, 2024-08-01:
Validé pour v2.1.3. J'archive l'issue.

closed 2024-08-01 by ronan-cariou