yguitton/CP-Seeker #43: User Guide

ronan-cariou, 2024-03-29:
Ajouter icône du user guide en haut à droite comme dans HaloSeeker. Mettre le user guide dans l’appli.

DamienCode404, 2024-04-03:
Ajout de l'icone :
[image]

Lorsque l'on survole avec son curseur, on a les informations sur la documentation (à changer en fonction du pdf) : 
[image]
Dans le fichier ui.R ligne 24 : `title = 'CP-Seeker 2.1.2 documentation', `

Lorsque l'on clique sur le bouton, on ouvre la documentation dans chromium : 
[image]

Dans le fichier regpaths.json, nous avons le nom exact du fichier, sinon il ne sera pas détecté par l'application : 
`"documentation_file": "documentation.pdf"`

Message d'erreur si le document n'est pas présent : 
`if (is.null(reg_paths$documentation_file)) stop('Missing documentation file in file reg_paths.json')
`

DamienCode404, 2024-04-04:
Changement du label pour ne pas avoir à préciser la version : 
[image]

Documentation de CP-Seeker :
[image]

jsaintvanne, 2024-04-23:
For me ok in https://github.com/yguitton/CP-Seeker/pull/47

ronan-cariou, 2024-08-01:
Validé pour v2.1.3. J'archive l'issue.

closed 2024-08-01 by ronan-cariou