yguitton/CP-Seeker #44: Infos survol de la souris

ronan-cariou, 2024-03-29:
Dans HaloSeeker, il est possible d’afficher des infos au survol de la souris au-dessus du plot interactif.
Peut-on afficher les infos de groupe homologue au survol d’une cellule ? Il s’agirait de l’intensité, du score, de la déviation et du statut inside/staddling/outside, quel que soit le filtre appliqué. Peut-être ajouter un info "Filter" In ou Out. Idem pour les standards.

DamienCode404, 2024-04-18:
**Piste pour aborder les modifications d'affichage**

Partie concernée par cet affichage est : `output$process_results_profile <- DT::renderDataTable`

- [ ] Ajouter une classe HTML grâce aux injections JS avec un nom différent pour chaque case.
- [ ] Récupérer (ou créer) la matrice qui aurait toutes les informations dont on a besoin pour l'affichage (non filtrée).
- [ ] Extraire les informations correspondantes à la case "hover" et les afficher avec du JS. 

_Trouver comment le fichier CPSeeker.js interagit avec l'affichage ainsi que les `htmlwidgets::JS`_

ronan-cariou, 2024-10-21:
Issue non abordée et dépriorisé. On ferme.
Ronan.

closed 2024-10-21 by ronan-cariou