yguitton/CP-Seeker #45: Parallelisation

ronan-cariou, 2024-03-29:
Mettre en place la parallélisation. Il s'agit de détecter le nombre de cœurs et de lancer en parallèle ce qu'il est possible de paralléliser.

1. Commencer par la déconvolution des chaque datafile.
2. Si ok, voire ensuite pour msConvert.

Est-ce applicable à la création des fichiers d'export, voire à l'affichage des graphiques ? Si oui, on en discute avant une nouvelle issue.

DamienCode404, 2024-04-04:
Test du temps d'exécution sur le même échantillon : jeu de données 20220611_072

Avant parallélisation :
[image]

Après parallélisation :
[image]

Gain moyen d'environ 35% mais le temps reste assez conséquent.

jsaintvanne, 2024-04-04:
Hello ! 
On y pensait hier avec Ronan aussi : attention à l'écriture dans la DB lors de la parallèlisation ! On pense que cela peut potentiellement poser problème d'avoir l'écriture de la déconvolution qui se fait en même temps... Cela est-il gérer par les packages de parallèlisation ? Toutes les données sont-elles bien sauvegardées ?

DamienCode404, 2024-04-08:
Je suis en train de tester CP-Seeker 2.0.0 (sans parallélisation) sur la machine PC retraitement 2. 
Le test s'effectue sur le jeu de 100 fichiers raw. 
Le but étant d'avoir un référentiel pour les prochains tests avec l'ajout de la parallélisation. 

Actuellement l'import des 100 fichiers a mis 1 h 30. 
Je suis en train de voir au niveau de l'import pour ajouter de la parallélisation.

En même temps, j'ai lancé l'étape de déconvolution avec Family = PCA / Adduct(s) = M+Cl.
En vue de la progression actuelle, je pense que l'on en a pour 4 h 10 environ. 
- 25 min pour 10% donc approximativement 250 min pour le temps total.
(Je vais éditer ce message lorsque le temps réel sera récupéré)

Demain, je vais voir le temps d'exécution de l'étape d'exportation en csv et xlsx. 

Une fois que ce premier test sera fini, je vais pouvoir comparer avec le nouveau code pour essayer de voir si on constate une réelle amélioration.

ronan-cariou, 2024-04-08:
Parfait. Juste une question sur la version. Pourquoi 2.0.0 alors que la version actuelle est 2.1.2 ?

ronan-cariou, 2024-04-08:
Ok alors on repart de zéro avec la version 2.1.2 stp. C'est la version à partir de laquelle tu travailles.
Ronan.

DamienCode404, 2024-04-08:
Ok pas de problèmes ! 
Je vais installer la version 2.1.2 de la branche main sur la machine PC retraitement 2. 
Je mets de côté ce que j'ai pour mon rapport, ça peut toujours être intéressant d'avoir une traçabilité de l'évolution des performances en fonction des versions du logiciel. Pour pouvoir argumenter la partie sur l'optimisation des performances.

ronan-cariou, 2024-04-08:
Dans la mesure su temps de process, assures-toi que personne d'autre ne sollicite le PC2, pour ne pas biaiser. Repères qui a une session ouverte et va poser la question dans le bocal. Nicolas a réservé la machine pour demain. Alternativement, je peux te laisser la place sur PC3.

DamienCode404, 2024-04-08:
Comment vérifier si une autre personne utilise sa session en même temps que moi ? 
Je ne pensais pas que l'on pouvait être à plusieurs sur un même PC retraitement.
Le problème étant que si je me connecte à un autre PC, je vais devoir à nouveau tout télécharger (version de CP-Seeker 2.1.2 et ma version). 
Je peux laisser la machine à Nicolas demain, je vais la réserver la machine PC retraitement 2 pour un autre jour.
Comment faire pour la réserver ?
Demain, je vais continuer d'ajouter de la parallélisation à l'import et à l'export et faire des tests moins longs en local.

ronan-cariou, 2024-04-09:
Ok pour l'approche. A terme, j'imagine que tu vas prendre plus que la moitié, mais pour les tests, ça me va.

DamienCode404, 2024-04-11:
Je suis sur la machine PC retraitement 2 et j'ai trouvé le moyen de voir l'utilisation des processeurs logiques individuellement. 
Il faut faire un clic droit dans le Task Manager > Change graph to > Logical processors

[image]

ronan-cariou, 2024-10-21:
Plusieurs pistes explorée mais mise œuvre mal maîtrisée et bénéfice flou.
Issue fermée sans implémentation.
Ronan.

closed 2024-10-21 by ronan-cariou