yguitton/CP-Seeker #52: C++

ronan-cariou, 2024-04-04:
Étudier le C++ écrit par Sébastien dans HaloSeeker voir comprendre comment il fonctionne et si on peut transposer à CP-Seeker pour donner un avantage.

DamienCode404, 2024-04-12:
Ajout de la librairie Rcpp "cppFuncs" avec une fonction de test multiplicationCpp : 

Exemple de l'appel de la fonction cpp par R : 

```
Restarting R session...

> library(cppFuncs)
> devtools::load_all(".")
ℹ Loading cppFuncs
> multiplicationCpp(3,5)
[1] 15
```

DamienCode404, 2024-04-19:
**Les erreurs avec la librairie Rcpp**

1. Lorsque je supprime le package Rcpp de HaloSeeker 2.0.3.3 : 

<p align="center">
[image]
</p>

_Donc, on peut conclure que le Rcpp chargé dans le dossier library de R est celui pris en compte lors du chargement. HaloSeeker fonctionne correctement avec sa librairie Rcpp version 1.0.5_

2. Lorsque je laisse le package Rcpp dans sa version 1.0.5 dans CP-Seeker (error.log) : 

```
Warning: Error in eval: function 'Rcpp_precious_remove' not provided by package 'Rcpp'
  59: eval
  58: eval
  56: source
  55: server [C:\Users\DRA\Desktop\CP-Seeker/server.R#69]
Error in eval(ei, envir) : 
  function 'Rcpp_precious_remove' not provided by package 'Rcpp'
In addition: Warning message:
In fun(libname, pkgname) :
  mzR has been built against a different Rcpp version (1.0.2)
than is installed on your system (1.0.5). This might lead to errors
when loading mzR. If you encounter such issues, please send a report,
including the output of sessionInfo() to the Bioc support forum at 
https://support.bioconductor.org/. For details see also
https://github.com/sneumann/mzR/wiki/mzR-Rcpp-compiler-linker-issue.
```
[error_2024-04-19-10h-14min.log](https://github.com/yguitton/CP-Seeker/files/15037869/error_2024-04-19-10h-14min.log)

_Selon le forum stackoverflow la solution serait de mettre à jour le package Rcpp à une version >=1.0.7._ 

_Source : https://stackoverflow.com/questions/68416435/rcpp-package-doesnt-include-rcpp-precious-remove_

3. La solution qui m'est venue à l'esprit était de mettre à jour la version de Rcpp dans le R-Portable. J'ai donc installé le package Rcpp 1.0.12, mais lorsque je lance CP-Seeker le chargement bloque au niveau du package MSnBase.

<p align="center">
[image]
</p>

[error_2024-04-19-10h-43min.log](https://github.com/yguitton/CP-Seeker/files/15037887/error_2024-04-19-10h-43min.log)

4. Problèmes avec les dépendances MSnBase et mzR.

```
In addition: Warning message:
In fun(libname, pkgname) :
  mzR has been built against a different Rcpp version (1.0.2)
than is installed on your system (1.0.5). This might lead to errors
when loading mzR. If you encounter such issues, please send a report,
including the output of sessionInfo() to the Bioc support forum at 
https://support.bioconductor.org/. For details see also
https://github.com/sneumann/mzR/wiki/mzR-Rcpp-compiler-linker-issue.
```

DamienCode404, 2024-04-30:
Après la mise à jour globale du R-Portables et de la librairie Rcpp, j'ai pu continuer à travailler sur le package.

La fonction "ppm_to_da_vector" de la librairie Rcpp "cppFuncs" est enfin fonctionnelle. 
Elle permet de remplacer le code faisant la conversion d'un vecteur de masses à une valeur de ppm en dalton.
Je pense pouvoir passer à des plus grosses fonctions à adapter au C++.

Ma librairie n'est pas chargée par le même système que HaloSeeker. Dans HaloSeeker les fonctions étaient exportées en un fichier dll puis lu par une fonction `dyn.load(dll_file)`. Actuellement, je charge simplement la librairie en l'ajoutant à la liste des autres packages. J'ai donc une copie de la librairie au niveau du R-Portable dans le dossier library. Je vais donc ajouter devant les fonctions C++ le nom de mon package pour les charger.

Exemple : 
`cppFuncs::ppm_to_da_vector(mass, ppm)`

ronan-cariou, 2024-10-21:
En conclusion, le C++ nécessite la MàJ de nombreux packages en cascade, ce qui s'est révélé être trop complexe.
Issue fermée sans implémentation.

closed 2024-10-21 by ronan-cariou