yguitton/CP-Seeker #53: Script Adam Cseresznye

ronan-cariou, 2024-04-04:
Etudier ce que faut ce script pour déterminer s'il y a de bonnes idées à intégrer à CP-Seeker.
https://github.com/adamcseresznye/CP-Seeker_post_processor

DamienCode404, 2024-04-05:
J'ai trouvé la documentation (lien ci-dessous) : 
https://github.com/adamcseresznye/CP-Seeker_post_processor/blob/main/notebook/CP_Seeker_Post_Processing_App.ipynb 

**Traduction de la documentation de CP-Seeker Post Processor :** 
L'application de post-traitement CP-Seeker automatise le traitement des données une fois que les données, y compris les niveaux de confiance et les zones de pic, sont obtenues à partir de CP-Seeker. Les processus manuels impliquent souvent des copier-coller répétitifs de données, introduisant des erreurs potentielles et de la lourdeur. Comme le montre @fig-example, la sortie de CP-Seeker pose des défis car les échantillons injectés sont répartis sur des feuilles Excel distinctes. De plus, chaque échantillon englobe deux dataframes sur le même onglet :

- un contenant les scores de confiance attribués par CP-Seeker aux congénères détectés,
- et l'autre détenant les zones de pointe réelles.

L'application de post-traitement CP-Seeker importe rapidement tous les ensembles de données à partir de différents onglets et restructure les trames de données, à l'instar de la fonction de fusion des pandas. Cette transformation organise soigneusement les congénères, en les empilant les uns sous les autres, plutôt que de les regrouper séparément par nombre de carbone et teneur en chlore. Par conséquent, tous les échantillons sont présentés côte à côte, éliminant ainsi le besoin d’onglets différents. De plus, l'application offre la possibilité de filtrer les congénères et les zones de pic en fonction d'intervalles de confiance. En d’autres termes, les congénères auxquels sont attribués des scores de confiance inférieurs peuvent être automatiquement exclus du traitement ultérieur. Une fois le traitement automatisé des données terminé, les données sont prêtes pour une analyse plus approfondie à l'aide du logiciel/flux de travail de votre choix.

Nous pensons que l'application de post-traitement CP-Seeker réduit considérablement le temps de préparation des données pour les étapes d'analyse ultérieures, améliorant ainsi la productivité et supprimant les erreurs potentielles associées au copier-coller manuel des données.

ronan-cariou, 2024-04-05:
Super.
Tu peux attacher un exemple de fichiers Excel input et output stp ? Est-ce ça fonctionne avec PCA uniquement, ou aussi avec PCO et PXA ? Ajouter aussi le csv output de CP-Seeker, pour je j'identifie bien la différence entre notre csv et son output à lui.
Merci,
Ronan.

DamienCode404, 2024-04-05:
Test du logiciel : 
[image]

Les fichiers excel qu'il utilise en exemple sont sous la forme de tableau avec des valeurs de scores, d'intensités et déviations.

**Score :** 
[image]
**Intensité :**
[image]
**Deviation :**
[image]

La sortie se présente comme une liste avec des valeurs les unes sous les autres avec le type d'élément Nombre de carbones et nombre de Chlores avec les scores d'intensité associé.
[image]

DamienCode404, 2024-04-08:
Les fichiers Excel qui ont été mis à disposition pour faire le "post processing". J'ai utilisé le test output 80% : 
[CPSeeker0.1_test_output_10_percent.xlsx](https://github.com/yguitton/CP-Seeker/files/14902202/CPSeeker0.1_test_output_10_percent.xlsx)
[CPSeeker0.1_test_output_50_percent.xlsx](https://github.com/yguitton/CP-Seeker/files/14902199/CPSeeker0.1_test_output_50_percent.xlsx)
[CPSeeker0.1_test_output_80_percent.xlsx](https://github.com/yguitton/CP-Seeker/files/14902201/CPSeeker0.1_test_output_80_percent.xlsx)

Le fichier de sortie en forme de liste au format csv, je ne vois pas sur le logiciel une autre forme d'export au format xlsx : 
[20240405_CPSeeker_post_processor.csv](https://github.com/yguitton/CP-Seeker/files/14902218/20240405_CPSeeker_post_processor.csv)

Le fichier de sortie csv de CP-Seeker : 
[1_2024-03-26.csv](https://github.com/yguitton/CP-Seeker/files/14902241/1_2024-03-26.csv)

ronan-cariou, 2024-04-08:
Vu avec Damien à l'instant, on n'a la même sentiment. Ce script est un travail inachevé et inexploitable pour une intégration à CP-Seeker. Nous avons déjà un export csv qui fait le job et prévoyons de créer les nouveaux modules de quanti et régression.
Par correction, contact a été pris avec Adam à ce sujet. En attendant, la présente issue est fermée.

ronan-cariou, 2024-04-09:
Pour info, réponse de Adam ci-dessous. Cette issue reste fermée et le script CP-Seeker_post_processor ne sera pas considéré.

_Hi Ronan,

In short, CP-Seeker Post-Processor is meant to clean up/reshape the output of CP-Seeker so that we can easily use the data in the following quantitation steps.

It is written to work with an older version of CP-Seeker (June 2022). In that version, the CP-Seeker's outputs were exported to an Excel file (xlsx) with separate sheets for each sample. Each sheet contained peak intensities and confidence levels for that specific sample.

For example, this is an example output from CP-Seeker, consider the peak intensities for sample 2309: 
[image]

 As you can see, the data is in a format where manipulation can be difficult. So, the CP-Seeker Post-Processor transforms this data into a more manageable "long" format. This involves converting each data frame into a single row and filtering intensities based on the confidence score (recommended to be set to 80 based on Tom's suggestion). Finally, we concatenate all the processed samples together. The resulting output places all peak areas (above 80% confidence) for all samples side-by-side on a single sheet, making it easier for further data processing. 

This is the output after these steps:  
 [image]

However, I noticed that version 2.1.2 of CP-Seeker might already have this functionality built-in, so my tool would be redundant for the newer versions. 

For now, at Antwerp, we continue to use the older version (1.6.x) of CP-Seeker because we noticed truncated peak areas in the output of version 2.1.2 (as previously mentioned in our email conversation). If a newer version exists that addresses this issue, we'd be very interested in trying it out. 

Kind regards,
Adam Cseresznye__