yguitton/CP-Seeker #54: Export csv, Nb chiffres après la virgule

ronan-cariou, 2024-04-09:
Dans l'export csv, la colonne "area" ne donne actuellement qu'un arrondi à l'unité, celle pour "deviation" un seul chiffre après la virgule. Coller à ce qui est fait pour l'export Excel, soit 6 chiffres après la virgule pour "area" et 2 chiffres après la virgule pour "deviation".

DamienCode404, 2024-04-09:
Voir un exemple de sortie csv ci-dessous : 
[1_2024-03-26.csv](https://github.com/yguitton/CP-Seeker/files/14915425/1_2024-03-26.csv)

DamienCode404, 2024-04-09:
Ah, mais du coup, je viens de voir que ce changement modifiait également l'affichage de la matrice de résultats :

Normalized Intensity (xE6)

[image]

Deviation (mDa)

[image]

Est-ce que je cherche à retraiter l'affichage de la matrice affichée dans CP-Seeker ou on valide comme ça ?

ronan-cariou, 2024-04-09:
Il faut conserver l'affichage de la matrice telle qu'elle était avant. Tu peux comparer le code entre l'export Excel et celui csv pour voir à quel niveau agir.
Ultérieurement, il y aura peut-être une issue sur l'ajout d'un paramètre d'affichage contrôlé par l'utilisateur pour définir le nombre de chiffres. Mais pas critique/prioritaire pour nous.

DamienCode404, 2024-04-16:
[image]

ronan-cariou, 2024-10-21:
Demande de modification bien prise en compte dans la v2.1.3.
Issue fermée.

closed 2024-10-21 by ronan-cariou