yguitton/CP-Seeker #59: Modifs lancement de v2.1.3

ronan-cariou, 2024-09-11:
Julien,

Fred Larvor est le premier a avoir installé la v2.1.3. Il a eu des soucis que j'ai confirmé. C'est pas très compliqué à résoudre je pense.

-Quand on lance l'exe pour la première fois, on a une fenêtre qui indique que le dossier Error_log a été crée (à la racine). Il faut cliquer sur ok pour la faire disparaitre En fait, l'utilisateur n'a pas besoin de savoir cela. Peut-on virer ce message ?

-Ensuite, l'installation s'arrête car il cherche le User Guide en v003 alors qu'on a remplacé par v004. Faire en sorte qu'il cherche bien la dernière version stp.

Merci,
Ronan.

DamienCode404, 2024-09-17:
Commit pour corriger l'affichage de la création du dossier Error_log : https://github.com/yguitton/CP-Seeker/commit/6ba9879e5db111253d9ce1fda91c71a01fd23b25
Commit pour corriger le lien vers la nouvelle documentation : https://github.com/yguitton/CP-Seeker/commit/d7d83a5179057bf68cd8d800fc0b4c10367c4e5b

ronan-cariou, 2024-10-21:
Je valide.
Passer ces deux commits sur la branche "main" et passer en v2.1.4. Je ne change pas la version du user guide pour ça.
Une fois fait, clore l'issue et m'informer.
Merci,
Ronan.

jsaintvanne, 2024-10-22:
PR made here https://github.com/yguitton/CP-Seeker/pull/61